FPGAwars/Alhambra-II-FPGA #3: Alhambra ADC example

drtrigon, 2018-10-27:
Is an ADC example for Alhambra 2 available?

I only found one for "MCP300x A/D controller (SPI master)" but as I understand is this the wrong chip and interface (SPI instead of I2C).

drtrigon, 2018-10-31:
Icezum Alhambra ADC: https://groups.google.com/forum/m/#!topic/fpga-wars-explorando-el-lado-libre/4_vHakcwXdQ

https://github.com/kaibran/printbot

Obijuan, 2018-10-31:
You can find a simple example here:

https://github.com/Obijuan/CTIF-Madrid-2018-FPGAs-Libres/tree/master/Colecciones/CTIF-MadridCap-6/examples/03-ADC

It just reads the AD channel 0 and show its value in the leds, at rate of 4Hz. The example is for the Alhambra I board, but the connection for the Alhambra II are the same

i am attaching here the example for the Alhambra II board

I am in the process of documenting everything. But in the meanwhile that example may help a little bit. The Alhambra II ADC converter is connected to the FPGA by the I2C bus

[05-adc0-leds.zip](https://github.com/FPGAwars/Alhambra-II-FPGA/files/2535521/05-adc0-leds.zip)

closed 2018-10-31 by Obijuan

drtrigon, 2018-10-31:
Thanks a lot! That is very helpful! 

(Now have to merge picorv32 with this... ;))

drtrigon, 2018-11-03:
[image]