FPGAwars/Alhambra-II-FPGA #4: Enhance specification with minimal power consumption

drtrigon, 2018-12-09:
I would like to know the minimal power consumption of the board an how I can minimize it. I did a very simple fast measurement and got around 60 to 70mA (with 4 LEDs on). Does that sound realistic?

I read that the TinyFPGA board goes down to "1.3 mA power when active", see https://www.crowdsupply.com/tinyfpga/tinyfpga-bx. Any chance that Alhambra 2 comes anywhere close to this?

EladioDM, 2018-12-10:
The 1.3 mA you mention refers to the oscillator only. Just the power LED may doubles that consumption. Anyway the static consumption of the whole board doesn´t go much higher, the FPGA and flash memory static current is under one milliamp.

In the Alhambra II, most of the power consumption comes from the peripherals around the FPGA. With 4 user LEDs and power LED on, peripherals power switch off and simple low freq circuit configured in the FPGA (like the demo bit stream), the Alhambra II draws about 35 mA: FTDI draws about 15 mA and LEDs 2.5 mA/LED. The remaining 10 mA goes to the ADC, the oscillator and the switching DCDC converter losses.

As you can see the board is not designed for low power but for ease to use and quick start up, with minimal or no addtional hardware.

May be it is not your case if you are pointing to low power applications, but consider it could be the opposite if you put the FPGA to work: Current drawn by the TinyFPGA using linear regulators could become higher than that drawn by the Alhambra II powered by switching regulators.

drtrigon, 2021-11-04:
Thanks for the detailed info!