FPGAwars/Alhambra-II-FPGA #8: Distance between pins 7 and 8

cesarab, 2020-01-15:
Hi.

Is there any reason why the distance between GPIO pins 7 and 8 is not multiple of 100 mils (2.54 mm)?

I wanted to create a hat for the FPGA, out of a regular perfboard, but it doesn't fit because of that.

The other side of the board does maintain the correct distance.

[image]

cesarab, 2020-01-15:
Nevermind. I can see that arduino has the same distance and this card keeps the Arduino form factor, so I gues that's the answer to my question.

closed 2020-01-15 by cesarab

Obijuan, 2020-01-16:
Yes, that is the reason: keeping the board compatible with Arduino Uno. All
the shields for the Arduino Uno can be used with the Alhambra II board