FPGAwars/Alhambra-II-FPGA #10: Speed-up memory access with PLL?

drtrigon, 2020-10-19:
I was working on https://github.com/drtrigon/fpgarduino-icestorm an have setup a picorv32 with peripherials in order to emulate/mimic an arduino uno board. During this I came to the point where I would have to implement `delayMicroseconds()` which is not possible due to the slow memory access.

Is it possible to speed up the SPI memory access, by making the clock faster using e.g. a PLL? Does a PLL block exist for icestudio? Does it work with the Alhambra II board?

drtrigon, 2020-10-26:
@Obijuan and @Jesus89 do you see a way of speeding-up SPI memory access (e.g. using a PLL)?

jcmadrioso, 2021-04-17:
I have been using PLL for speeding up clock, but only with icestorm. I'm trying to use it in IceStudio via module instantiation.

drtrigon, 2021-11-04:
@jcmadrioso: Can you share the code?