FPGAwars/Alhambra-II-FPGA #11: VHDL Support

pedro-javierf, 2021-09-27:
I was wondering what learning resources are available regarding the programming of the iCE40HX4K in the Alhambra-II using the VHDL language rather than Verilog/SystemVerilog. The reasoning for this is, lots of Spanish universities teach VHDL and not Verilog, thus it'd be great for students to be able to leverage their learnt FPGA skills using this board.

As far as I'm concerned, the only toolchain that supports this is the "official" Lattice iCEcube2. But how can a bitmap generated with this privative software (or others) be programmed into the board over USB?

Werni2A, 2021-11-27:
Shouldn't Lattice's toolchain be able to also program the Alhambra Board? I can't imagine that this does not work. Anyhow an open-source approach is provided by iceprog from the [icestorm](https://github.com/YosysHQ/icestorm) or [openFPGAloader](https://github.com/trabucayre/openFPGALoader).

A short example below shows how to use both tools to program the ice40 FPGA (both were tested on my board). Still you have to adjust `PWD` for your current working directory and `PROJ` with your bitstream project name.

```
DOCKERARGS=run --rm -v /$(PWD)://wrk -w //wrk
ICEPROG=docker $(DOCKERARGS) --device /dev/bus/usb gcr.io/hdl-containers/prog:latest iceprog
OPENFPGALOADER=docker $(DOCKERARGS) --device /dev/bus/usb gcr.io/hdl-containers/openfpgaloader:latest openFPGALoader

$(ICEPROG) -d i:0x0403:0x6010:0 build/$(PROJ).bin
$(OPENFPGALOADER) -b ice40_generic build/$(PROJ).bin
```

Werni2A, 2021-11-28:
A simple design can be found at [my repo](https://github.com/Werni2A/Valhalla-II) that should get you started with a pure OSS VHDL flow.

pedro-javierf, 2021-11-28:
Thanks for the OSS flow @Werni2A ! I will definitely look into it. I had already found and documented the lattice VHDL flow [here ](https://pedrojavier.com/2021/10/11/tutorial-how-to-use-vhdl-with-the-alhambra-ii-fpga/) but this is what I really wanted :)