FPGAwars/Alhambra-II-FPGA #13: Please program the 'serial code' field of the Alhambra-ii boards you ship

zapta, 2025-05-15:
Following is the output of ``openFPGALoader --list`` with two FPGA boards, alhambra-ii and upduino.  Notice that the alhambra doesn't have the serial code programed in the expected field while the upduino does.  Thsi field allows the openFPGALoder programmer to distinguish between multiple similar boards that are connected to the host.

This will also allow apio to distinguish between boards when programming using openFPGALoader:  ``apio upload --serial FT94RQ8V``.

```
Bus device vid:pid       probe type      manufacturer serial               product
000 001    0x0403:0x6010 FTDI2232        AlhambraBits none                 Alhambra II v1.0A - B09-335
001 001    0x0403:0x6010 FTDI2232        tinyVision   FT94RQ8V             UPduino v3.1c
```

@cavearr, FYI, related to https://github.com/FPGAwars/apio/issues/596

cavearr, 2025-05-15:
Very interesting @zapta ! i'm reply you at the other issue

cavearr, 2025-05-15:
@zapta ,Apparently I was mistaken, and openfpgaloader doesn't save to SRAM; it actually tries and ends up saving to flash. It appeared as Not Tested in the documentation, and my initial tests gave me the impression that it worked.

Sorry for the misunderstanding. I'll investigate what you're saying to see if we can find a solution.

In any case, using openfpgaloader now or in the future is more than promising.

cavearr, 2025-05-15:
yes!, i'll talking with them now., tell you soon 

@zapta

zapta, 2025-05-18:
Hi @cavearr, the new ``apio drivers list ftdi`` command now prints also the serial number field so I don't think it's necessary anymore to include the board's serial number (``B09-335`` in the example below) in the product description field.

```
$ apio drivers list ftdi

┌──────────┬───────────────┬─────────────────────────────┬──────┬──────┬──────────┬─────┬────────┐
│ TYPE     │ MANUFACTURER  │ DESCRIPTION                 │ VID  │ PID  │ SERIAL   │ BUS │ DEVICE │
├──────────┼───────────────┼─────────────────────────────┼──────┼──────┼──────────┼─────┼────────┤
│ FTDI2232 │ tinyVision.ai │ UPduino v3.1c               │ 0403 │ 6010 │ FT94RQ8V │  1  │   1    │
├──────────┼───────────────┼─────────────────────────────┼──────┼──────┼──────────┼─────┼────────┤
│ FTDI2232 │ AlhambraBits  │ Alhambra II v1.0A - B09-335 │ 0403 │ 6010 │          │  0  │   1    │
└──────────┴───────────────┴─────────────────────────────┴──────┴──────┴──────────┴─────┴────────┘
Found 2 devices
```

cavearr, 2025-05-18:
Hi @zapta ! i'm talking with Alhambrabits about it and they listen to us. This week they send us an ftdi template to test it an if it works they put in all of new Alhambra boards.

I'm exploring ftdi in deep to look for a solution and other fpga friend give me a possible solution, i'm testing it and tell you in the next days.

zapta, 2025-05-18:
Thanks @cavearr, that will be very useful.  I don't know if its possible but it if the FTDI chips come from FTDI with a serial number, it would be nice to leave that number as is.

zapta, 2025-05-19:
@cavearr, here is another motivation for having standard serial number for Alhambra, when Mac OSX maps FTDI channels to serial ports, it includes the serial number in the port name if available, for example ``/dev/cu.usbserial-FT94RQ8V0`` for serial number ``FT94RQ8V``. This provides one more way to distinguish between devices if more than one if connected.

```
$ apio drivers list serial

SERIAL Devices                                                                                          
┌───────────────┬─────────────────────────────┬──────┬──────┬────────────┬─────────────────────────────┐
│ MANUFACTURER  │ DESCRIPTION                 │ VID  │ PID  │ SERIAL-NUM │ PORT                        │
├───────────────┼─────────────────────────────┼──────┼──────┼────────────┼─────────────────────────────┤
│ tinyVision.ai │ UPduino v3.1c               │ 0403 │ 6010 │ FT94RQ8V   │ /dev/cu.usbserial-FT94RQ8V0 │
├───────────────┼─────────────────────────────┼──────┼──────┼────────────┼─────────────────────────────┤
│ tinyVision.ai │ UPduino v3.1c               │ 0403 │ 6010 │ FT94RQ8V   │ /dev/cu.usbserial-FT94RQ8V1 │
├───────────────┼─────────────────────────────┼──────┼──────┼────────────┼─────────────────────────────┤
│ AlhambraBits  │ Alhambra II v1.0A - B09-335 │ 0403 │ 6010 │            │ /dev/cu.usbserial-1201      │
├───────────────┼─────────────────────────────┼──────┼──────┼────────────┼─────────────────────────────┤
│ AlhambraBits  │ Alhambra II v1.0A - B09-335 │ 0403 │ 6010 │            │ /dev/cu.usbserial-1200      │
└───────────────┴─────────────────────────────┴──────┴──────┴────────────┴─────────────────────────────┘
Found 4 devices
```

cavearr, 2025-05-21:
Fantastic @zapta!, I'm waiting for Alhambrabits to send me the Eprom programming templates and I'm investigating how to change it with a small tool of ours. As soon as I have all the info and consolidate it, I'll share it with you.

zapta, 2025-05-21:
@cavearr, FYI, according to the ftdi standard programming tool, the total size of manufacturer, product, and serial string should be <= 46, and max serial number length is 16.

[image]